Opentrons/opentrons #2047: Run log out of sync with actions after pause

rvinzent, 2018-08-13:
## overview

The real time run log becomes out of sync with the actions of the robot after pausing and unpausing.

## current behavior

The app's log scrolling and highlighting action does not match with the robot's current actions after a pause.

## steps to reproduce

Run any protocol of moderate to long length. Pause in the middle of a run, wait for a minute, then unpause. Repeat if necessary. Actions should be out of sync with the run log after unpausing.

App version `3.2.0`
Robot server version `3.2.0`

## expected behavior

Run log should stay consistent with the actions of the robot.

mcous, 2018-08-13:
Oops, this is a known issue that should have already been ticketed. Thanks for the report! Run log also may or may not be out of sync _before_ pause, too. Fun times

simonchih, 2019-03-22:
App Pause may make magnetic module won't do engage or disengage with version 3.6.5.

Matt-Zwimpfer, 2021-06-03:
LMH - known about and should be resolved in Protocol Engine

mcous, 2022-03-04:
It gives me great satisfaction to finally mark this issue as resolved in v5

closed 2022-03-04 by mcous