Opentrons/opentrons #2054: Robot Returning Incorrect Name

umbhau, 2018-08-13:
## Expected Behavior
- API should be returning robot ID from config (so that the App can display it)

## Observed Behavior
- API is returning longer string name

## Steps to recreate
- Connect to any robot
- Observe 'Robot Name' in 'Information' card is reported as:
"opentrons-OT2P20180314A02", or "Opentrons on Brians-MacBook-Pro-2"
rather than as the serial number/id:
"OT2P20180314A02"

umbhau, 2018-08-13:
Not very familiar with how id/name (?) are configured as data / returned via API; filing this bug on behalf of @andySigler who reports the following:

> the #1 issue the SZ team kept reminding me of is that the App doesn't show the robot name as an ID anymore. Now it shows the random English words resin first assigns it like `Opentrons-still-paper`, but no longer shows the ID they then rename it to like `OT2P20180812A01`. They say this is super duper annoying because they don't know which robot is which

With the following clarification from @mcous:
> At some point the API appears to have stopped being able to pick up the ID from configuration and so it’s reporting its name as the words name to the app, which is what gets displayed

mcous, 2018-08-17:
Removing the `opentrons-` name prefix is a **breaking change** that will prevent older versions of the app from finding newer versions of the robot (that don't have the prefix) on the network

btmorr, 2019-03-25:
@mcous Should we close this in light of #2357? Seems like this is probably something we're not going to change

mcous, 2019-03-25:
@btmorr I would prefer we don't add `opentrons-` automatically to the advertised name unless we have a good reason to do so.

@sfoster1 in [this comment](https://github.com/Opentrons/buildroot/pull/9/files#r266095875) you mentioned something about advertising our capabilities more specifically than `tcp`; do you think that's something likely to ship with the new system?

mcous, 2019-07-26:
This behavior has been changed in the new buildroot-based system; closing

closed 2019-07-26 by mcous