Opentrons/opentrons #2059: Feature request: ability to update custom container definitions

rvinzent, 2018-08-14:
## overview

Updating a container definition is currently not supported. There is also no way to remove a custom container definition through the API. As a user, I would like the labware database to be updated with the new definition when I run `labware.create` a second time on a container with the same name, or use a separate `update` or `delete` method to better interact with the custom labware.

## current behavior

Can only create containers through the API. Cannot update definitions because it violates the unique constraint on the `name` column. Cannot delete container definitions to clean up the database either.

The only work around is to query the database manually.

## expected behavior

Basic CRUD operations supported by the API for interacting with the custom labware database.

buffavaholic, 2018-08-17:
I also think it would be useful to be able to overwrite a container definition. Possibly this could be done by keeping the to be deprecated `opentrons.data_storage.database.overwrite_container()` function not as a calibration tool but to actually overwrite custom containers

laserson, 2018-12-18:
+1. I created a custom container incorrectly and now my preferred identifier is "used up". Perhaps you could post a workaround for this to manually remove the container.

mcous, 2018-12-18:
While drastic, the "Robot" > [Name] > "Advanced Settings" > "Factory Reset" > [x] "Labware Calibration" option will delete _all_ previously created custom labware in addition to modifications made to existing labware by runtime calibration. After labware reset+restart, the robot will take longer to boot than normal (several minutes) as it re-creates its labware database from the base definitions.

All our engineering energy is currently focused on removing and replacing the labware database with a more sane and reliable management system, so unfortunately additions to the current database API are unlikely. I'll make sure this stuff is on the radar as we continue to push documenation improvements, though. Sorry for the inconvenience and we appreciate your patience

mcous, 2019-08-29:
We are completely re-doing how the robot deals with labware. We have released the first stage of this project (all new definitions, deprecation of the on-board labware SQL database).

To track the custom labware aspect of this project, I'd keep an eye on this milestone: https://github.com/Opentrons/opentrons/milestone/109

closed 2019-08-29 by mcous