Opentrons/opentrons #2079: Feature: ability to play mp3 files

arachnologist, 2018-08-17:
Feature Request

Our lab really wants to take advantage of the OT-2's speakers and code protocols to play an mp3 file after coding the robot to pause so we can create a kind of alarm that lets us know the robot needs a human to do something and resume the protocol. When I contacted the support team, they said I should be able to use mpg123, but the API doesn't recognize that module yet? Sorry if this is the wrong place for this kind of issue, but I've asked the support team so many times about this and never get a response.
We've tried making simple beeps with 'os.system' and 'winsound' which is accepted by the API, but no sound is produced when the code runs.

App version: 3.2.0; firmware: edge-6168d32; software: 3.2.0

rvinzent, 2018-08-17:
Hi @mcous,

Can there be some support for loading a callable into the actions to be executed at runtime? Then we can play sounds, update databases/tracking spreadsheets, etc. when the robot is running versus when the script is loaded.

mcous, 2018-08-20:
@arachnologist: `mpg123` is a command line application, not a Python module. If you would like to call it, you should use the [subprocess](https://docs.python.org/3.6/library/subprocess.html) Python module. I'm not really a Python developer myself, so I'm not super familiar with how `subprocess` works, but Python docs + Google + Stack Overflow should be helpful in this regard.

@rvinzent if I'm understanding you correctly, you'd like to take some arbitrary action during the protocol run but not protocol simulation at load? If that's the case, I think the `robot.is_simulating` method can get you what you need.

```py
import subprocess

# ...

# play a sound when the run gets here
# note: we're not in love with the robot.is_simulating API and it may be removed in a
# future major version bump, hence this assert guard
assert callable(getattr(robot, 'is_simulating', None)), 'robot.is_simulating not callable'

if robot.is_simulating() is false:
    # not sure if this is correct; see above caveat about not being a python dev
    subprocess.run(
        'mpg123 ./path/to/file/on/rasp-pi.mp3',
        shell=True, stdout=subprocess.PIPE, stderr=subprocess.PIPE
    )

# ...
```

arachnologist, 2018-08-22:
@mcous: your code for the simulation gate works as long as you make sure the boolean response is entered as a string (in your case, false should be 'false'). The subprocess code doesn't work, but I'm also not a python developer, and the subprocess module itself is still a mystery to me, so I can't help with that.

mcous, 2018-08-22:
@arachnologist that was my bad, I forgot that python uses `True` and `False`, not `true` and `false`. I've updated the snippet in my post accordingly. `if robot.is_simulating() is 'false'` is incorrect and would never evaluate

arachnologist, 2018-08-30:
Finally uploaded our sound file onto the raspberry pi. This is the line of python code that worked for us:
>import subprocess
>subprocess.call('mpg123 /hey_listen.mp3', shell=True)

"hey_listen.mp3" is the name of the sound file being accessed from the raspberry pi's files
Hopefully helpful to anyone still trying to use mp3 files in their protocols.

sfoster1, 2019-12-13:
That's awesome! Glad it worked out for you.

closed 2019-12-13 by sfoster1