Opentrons/opentrons #2095: App: container doesn't calibrate if container has label

buffavaholic, 2018-08-21:
## overview

Bug: Protocols that load containers with a label does not save the calibration and move to the next container during calibration. 

## current behavior

If I use the OT logo protocol (http://protocols.opentrons.com/protocol/Opentrons_Logo) (with a p300 and tube rack) the calibration procedure is completed as normal. But if I change the container loading to utilize the label parameter then after doing the calibration for the tiprack, and clicking 'move to labware' it opens the calibration window and I can jog but when I press 'save calibration' the calibration window closes briefly but then re-opens again. The only way to get out is by hitting the 'back' button but the container isn't calibrated

## steps to reproduce

Use the attached protocol in the app and complete the calibration
[Opentrons_Logo_wNames.txt](https://github.com/Opentrons/opentrons/files/2307862/Opentrons_Logo_wNames.txt)

## expected behavior

It should complete the calibration similar to when labels aren't use on the labware

mcous, 2018-08-22:
Unable to reproduce on current `edge` with virtual smoothie. Will continue to investigate with a real robot

mcous, 2018-08-22:
@buffavaholic I'm unable to reproduce the described behavior with the protocol provided. Are you on the latest stable app + robot API + robot firmware versions?

* App: `3.2.0` (More > App > Information > Software Version)
* API: `3.2.0` (Robot  > {your robot's name} > Information > Server version)
* Firmware: `edge-6168d32` (Robot  > {your robot's name} > Information > Firmware version)

Also, what OS are you running?

buffavaholic, 2018-08-22:
@mcous I was using an edge from a couple weeks ago. I just updated and don't see the issue any more. Thanks! (Now works on App 3.2.0 API 3.2.0 and firmware edge-6168d32)

closed 2018-08-22 by buffavaholic