Opentrons/opentrons #2096: Feature: temporary well locations for colony picking

buffavaholic, 2018-08-21:
## Overview

Feature request: (API)
Given a list of the (x,y) coordinates of locations to visit on a SBS shaped agar plate, I would like to add wells to a plate to be able to point the pipette to those locations, but I do not want the locations saved permanently so that the same base container can be re-used/used in multiple slots with different labels. I have found this possible by adding wells to a plate after loading but I have found that the Container ordering and robot.poses are not updated with the added well. 

## Feature base requirements

- Use a single plate definition for bacterial colonies
- Add wells to a given instance of a plate using a list of (x,y) positions (relative to the SBS shape lower left hand corner) 
- Utilize the same liquid handling function as a normally defined well
- Added well locations not saved when defined

## Proposed approach

First I created a custom labware with 1 well ('A1") which is localized at (0,0)
`labware.create('colonyBase',(1,1),(0,0),5,10,volume=10)`

Then in the protocol you would load this container and add wells to it. But I have found that in addition to using Container.add to add a new well the well needs to be added to the Container.ordering and to the pose_tracker. See protocol -[testColPick.txt](https://github.com/Opentrons/opentrons/files/2307968/testColPick.txt)

I think that basically this all could be wrapped into a method that takes the container and a list of (x,y) locations and add the wells this way. Possibly something like this-[possibleColonyMethod.txt](https://github.com/Opentrons/opentrons/files/2308047/possibleColonyMethod.txt)

(I have tested the protocol and seen that it works but not the proposed method)

## Possible issues/conflicts
- Unsure if other well settings are not set during Container.add that some other methods use
- Unsure if adding wells after the fact messes up other processes
- Currently the app doesn't calibrate containers that aren't used with some liquid handling method, whereas colony picking would really just use move_to. I added an aspirate at well 'A1' just to make it calibrate. 
- Maybe the addition of the ordering and to the position tracker should be rolled into Containers.add so that those things are done by default but I don't know how that would work with how they are set up now. 
- Keeping these functions could be affected in the future if the labware definitions get saved more than just when one is added.

buffavaholic, 2018-08-22:
I realized i was on an old edge and I just updated my firmware and it appears that adding the tracking is not necessary, but the addition to the pose_tracker is still required

mcous, 2020-01-16:
Labware creation (and the Protocol API itself) has changed significantly since this ticket was opened; I'd definitely recommend checking out the [Labware Creator](https://labware.opentrons.com/create) and [Protocol API v2](https://docs.opentrons.com) to see if they fit your needs a little better

For now, I'm going to close this ticket in favor of the slightly newer and more generic #3441 so we don't lose track of colony picking as a feature request

closed 2020-01-16 by mcous