Opentrons/opentrons #2120: For the p10 allow user to input 1 decimal place

howisthisnamenottakenyet, 2018-08-23:
As a user with a P10 I'll want to be able to input volumes such as '2.5'.

## Acceptance Criteria
- [ ] Allow users to input a "." into the volume field if a step uses a P10
- [ ] Allow users to input 1 number past the decimal place
- [ ] If a user fills out the P10 volume with a decimal field and then changes the pipette to a non-P10 one of two things can happen. Whichever makes most sense given how we're building things.
(1) The volume field clears
(2) Digits and the decimal point clear but digits before the decimal point stay

howisthisnamenottakenyet, 2018-08-23:
@IanLondon @b-cooper @umbhau 

I know there were a lot of opinions on decimal points + text fields. Does this proposal make sense for P10s?

umbhau, 2018-08-23:
@howisthisnamenottakenyet slight preference for keeping it the same across the board, just for simplicity's sake. but extra significant digit makes sense.

howisthisnamenottakenyet, 2018-08-23:
@umbhau At the level of a P10 yeah. One of our testers actually referred to this as a bug.

IanLondon, 2018-08-24:
Seems good to me, matches what is on a manual P10

commit 827f3ee5 (2018-09-12): feat(protocol-designer): allow tenths of µl pipette volumes (#2222)

Allow users to input volumes in step forms with precision down to the tenth of a µl.

Closes #2120