Opentrons/opentrons #2129: Protocol Info

umbhau, 2018-08-26:
PLACEHOLDER

As a Run App user, I would like to be able to see basic info about my protocol when I upload it to the Run App, so that I know what to expect from the run.

## Acceptance Criteria
- TBD

## Design
- @pantslakz

pantslakz, 2018-08-28:
Designs for review:

Zeplin > Run App Screens > Tagged "File Info"

howisthisnamenottakenyet, 2018-08-28:
@umbhau @pantslakz  I'm not sure what the user would get out of having the protocol type there. However we should be displaying PD version it was made with if it's a JSON file.

pantslakz, 2018-08-28:
I think for us it may seem obvious if something is python or Json, but there could be cases where users have the extension turned off on their OS file options, and unsure of what the file type is? 
also I added it for consistency between file types

pantslakz, 2018-09-05:
@Kadee80 @mcous please note we'll need to add "modules" to this popup:

[image]

mcous, 2018-09-05:
@pantslakz even if the user has their "display extensions" option turned off, the files themselves still have extensions. We'll be operating on the technical assumption that if a file has a `.json` extension, then it is a PD protocol, and if it's anything else, we'll try to upload it as a Python protocol (I've seen support cases where users save their Python protocols as `.txt` instead of `.py`).

A message somewhere communicating this logic to the user might be helpful; maybe it would make sense to put it in the failed protocol parse/upload screen

commit d64153ef (2018-09-14): feat(app): Populate FileInfo page with JSON protocol metadata

Closes #2129
commit bc367c45 (2018-09-14): feat(app): Populate FileInfo page with JSON protocol metadata

Closes #2129
PR #2278 (2018-09-14): feat(app): Populate FileInfo page with JSON protocol metadata
## overview

Closes #2129

This PR adds selectors for JSON protocol metadata and wires them up into the File Info page. It also tweaks the error message to show up on the file info page (because protocol metadata would be useful to debug a JSON protocol that was valid JSON but failed during simulation).

[image]

## changelog

- feat(app): Populate FileInfo page with JSON protocol metadata 

## review requests

Please test this out with:

- [x] Good JSON protocol
- [x] Bad JSON protocol
- [x] Good Python protocol
- [x] Bad Python pr [... cut]
commit 995038a4 (2018-09-14): feat(app): Populate FileInfo page with JSON protocol metadata (#2278)

Closes #2129
closed 2018-09-14 by mcous