Opentrons/opentrons #2133: Calibrate Pipette Mount: Confirm Success and Exit

umbhau, 2018-08-26:
As a robot user whose robot completed pipette mount calibration tip probe, I would like to be informed of its success and given the option to exit or change another pipette.

## Acceptance Criteria
- After robot successfully completes tip probe, user informed pipette mount calibration is complete, and that their robot's calibration has been update
- User given option to exit and return to robot settings page
- User given option to change another pipette

## Design
- TBD

## Implementation Notes
- Note this story does not cover failure case. App/robot does not handle failure gracefully today; failure case will be covered in a separate user story, but should be implemented independently.

pantslakz, 2018-09-11:
in current implantation - we do not allow user option to change another pipette. user must exit and select change again from robot settings page.

umbhau, 2018-09-11:
@pantslakz updated to only cover exit

pantslakz, 2018-09-18:
[image]

closed 2021-04-08 by sfoster1