Opentrons/opentrons #2134: Network Connectivity: 802.1x Usernames

umbhau, 2018-08-26:
As a robot user, I would like to be able to connect to my robot to 802.1x wifi networks that require a username for authentication, so that I can access Opentrons' updates and support.

## Acceptance Criteria
- Run App displays username field for authentication when appropriate (?)
- User able to authenticate to 802.1x network with username and password

## Design
- TBD

## Implementation Notes
- Is it possible to determine when username is required?
- Will likely split into RA and API stories

mcous, 2018-09-17:
Ping @umbhau and @pantslakz - following up on questions from above:

> Does the robot know what type of authentication protocol it is dealing with?

The robot knows the difference between "no security", "standard username/password", and enterprise. Once we're in the "enterprise" bucket, there are several (5 at the time or writing) we can support, but the robot has no way of telling us which of those 5 a network uses.

> What auth protocols should we allow the user to select?

We have enough information to drop the user to:

- No password box (or disabled box) for networks with no security
- Password box for `WPA2` networks
    - After talking to @sfoster1, for now we can safely ignore `WEP` (deprecated 2004), `WPA1` (deprecated 2006), and anything else
- API-driven enterprise selection box
    - #2283 adds an endpoint that returns enterprise security types and a list of required and optional fields for each type
    - A field has a `name`, a `displayName`, a `required` flag, and a `type`
    - There are three types of field
        - Normal text (e.g. "Username")
        - Password (e.g. "Password")
        - Certificate file ID (e.g. "Private Key File") **see below**

**Certificate files**

Some enterprise authentication mechanisms require one or more certificate files. Selecting a certificate file for a given input is a multi-step process:

1. Get list of certificates already on the robot
    - If certificate is already loaded, populate field with the ID and you're done
2. If not already on robot, upload the file to the server
3. Server will return an ID, populate certificate field with the ID

If we wanted to, we could get away with not building a certificate management flow and just upload the files every single time. This would probably be subpar from a UX standpoint but is technically feasible.

umbhau, 2018-10-03:
Closing in favor of: 
Manage Robot Connection: Select Type of 802.1x Authentication #2412 
Manage Robot Connection: Enter Authentication Information #2413
Manage Robot Connection: Certificate Authentication #2415
Manage Robot Connection: Required Fields #2416

closed 2018-10-03 by umbhau