Opentrons/opentrons #2199: Add PCR strip to PD labware options

howisthisnamenottakenyet, 2018-09-06:
## Notes
- There are multiple sizes. Which do we want?

pantslakz, 2018-10-24:
@IanLondon @Laura-Danielle how does PCR strips work in python files? Does user place a 96 well plate first, then pcr strips on top? what does the lab ware definition encompass? is it like a tube rack?

IanLondon, 2018-10-24:
@pantslakz 

Similar to a how a tube rack definition has the rack filled with tubes as one definition, the PCR strip defs include all strips you can fit with the PCR rack together. I guess technically a PCR strip holder _is_ a tube rack.

AFAIK you usually wouldn't put PCR strips in a 96 well plate, you put them in a PCR strip holder:

[image]

It looks like the only definitions we have at the moment are:
* `PCR-strip-tall` (96 tubes. is it actually "tall"? Volume: 0.28mL)
* `alum-block-pcr-strips` (only has 2 strips = 16 tubes, why? Volume: 0.28mL)
* `opentrons-aluminum-block-PCR-strips-200ul` (96 tubes. It doesn't seem tall enough to include the aluminum block...? Volume: 0.3mL)

So we don't have any 0.5mL PCR tube defs - unless they just haven't been migrated -- @Laura-Danielle do you know?

IanLondon, 2018-10-30:
Closed by #2583 (which includes `PCR-strip-tall` and `opentrons-aluminum-block-PCR-strips-200ul`)

closed 2018-10-30 by IanLondon