Opentrons/opentrons #2200: Some pipettes struggle to pick up tips

howisthisnamenottakenyet, 2018-09-06:
As a user I may have pipettes that need extra presses to pick up tips. This option is necessary for me in PD.

## Acceptance criteria
- [ ] User should be able to specify the number of presses each pipette should execute when trying to pick up tips

## Design

pantslakz, 2018-09-06:
Im not convinced the best solution to this is in PD. I think this could be something to consider in Run App - would prefer we collect info on status of this with help of product & support team before jumping to solution @howisthisnamenottakenyet

howisthisnamenottakenyet, 2018-09-06:
@pantslakz updated tags and description

IanLondon, 2019-01-11:
Will be addressed in user-editable pipette config project, via Run App!