Opentrons/opentrons #2279: AKC timeout error for JSON files

howisthisnamenottakenyet, 2018-09-14:
Sometimes i get this error trying to import a JSON file into the run app. If I keep trying eventually it does successfully import though. 
[image]

mcous, 2018-09-14:
@howisthisnamenottakenyet do you have a protocol handy where this happens often?

@IanLondon my best guess here is that something about the JSON handling in `session.py` is blocking the RPC server until it's done, which could prevent the ack from coming back within the timeout (currently set to 10 seconds)

mcous, 2019-08-15:
Haven't seen this happen in a while, and assume it was related to some of the old simulation bugs. Regardless, will no longer be an issue with RPC removed, so closing this issue

closed 2019-08-15 by mcous

ethanfjones, 2019-09-04:
@mcous I have a user experiencing this at the moment. I am going to reopen the issue. 

Every time I upload this protocol I lose connection with the robot. 

[AKC_timeout.zip]
(https://github.com/Opentrons/opentrons/files/3575714/AKC_timeout.json.zip)