Opentrons/opentrons #2292: Labware Definition Issue: 96-pcr Aluminum Block

umbhau, 2018-09-17:
Labware definition for 96-pcr aluminum block is slightly off.

## Expected:
- Pipette should move to center of well during labware calibration

## Observed:
- Pipette moves too far back and left during labware calibration (not above well at all)

##: Steps to Recreate
- Perform robot deck calibration
- Load a protocol with 96-pcr aluminum block
- Configure deck with aluminum block on temp deck and pcr plate on aluminum block
- Perform labware calibration and observe pipette move outside of well

Note this was only observed with the 96-pcr (other aluminum blocks are better)

umbhau, 2018-09-17:
FYI @Laura-Danielle @btmorr

2 of 2 labware issues @mcous and I found while testing 3.4.0 labware updates. Let's consider this high priority and blocking 3.4.0 production release (and therefore use of new aluminum blocks)

btmorr, 2018-09-20:
Strips for reference:

[image]

btmorr, 2018-09-20:
It also appears that (in the existing labware definition system) we need to offset from the x and y of the machine drawings by -1mm and -6mm respectively, based on practical experimentation on multiple robots.

commit 3e5177cc (2018-09-20): fix(api): Update the aluminum block definitions to match drawings and experiments

Closes #2292
commit f6024cfc (2018-09-20): fix(api): Update the aluminum block definitions to match drawings and experiments

Closes #2292
PR #2342 (2018-09-21): fix(api): Update the aluminum block definitions to match drawings
## overview

Update the aluminum block definitions to match drawings and experiments (see associated issue for details). Closes #2292

## review requests

- [ ] Verify that the robot has a correct deck calibration (can move accurately to calibration crosses)
- [ ] Go to top and bottom of first and last wells for `"opentrons-aluminum-block-96-PCR-plate"`
- [ ] Go to top and bottom of first and last wells for `"opentrons-aluminum-block-PCR-strips-200ul"`
commit 4c1e4c23 (2018-09-21): fix(api): Update the aluminum block definitions to match drawings (#2342)

* fix(api): Update the aluminum block definitions to match drawings and experiments
* Rename definition in shared-data to match rename in default_containers.json and fix z height
* Closes #2292
closed 2018-09-21 by btmorr