Opentrons/opentrons #2334: PD Deck Setup: Sticky Start and End in Protocol Timeline

umbhau, 2018-09-20:
As a protocol designer, I always want to be able to see the beginning and end of my protocol timeline in the protocol timeline sidebar.

## Acceptance Criteria
- 'Edit Protocol Deck Setup' is sticky at the top of the sidebar
- 'Result' is sticky at the bottom of the sidebar
- Change name of 'Starting Deck State' to 'Edit Protocol Deck Setup'

umbhau, 2018-09-26:
on hold pending open design discussion w/ @pantslakz

btmorr, 2019-01-04:
I'm pretty sure this is not a useful ticket anymore. Closing. @umbhau , @pantslakz , or @howisthisnamenottakenyet if there is something to be done related to this, please either re-open this ticket with details, or open a new ticket

closed 2019-01-04 by btmorr