Opentrons/opentrons #2337: Protocol Library: Markdown Rendering Issues

umbhau, 2018-09-20:
Expected Behavior:
- Ingredient List written in markdown should render on protocol library page

Observed Behavior
- Ingredient List in markdown not rendering on protocol library page

IanLondon, 2018-09-26:
may be fixed just by changing the number of `#`s in the README, a real solution is probably gonna be costly with things as they are

btmorr, 2019-01-04:
@umbhau is this still relevant?

btmorr, 2019-06-25:
Closing this as stale. Please open a new ticket if needed. (also note that the example snippet doesn't have any empty lines between the header and the bulleted list, which I'm pretty sure is not valid for a list in standard markdown--that might also be causing problems)

closed 2019-06-25 by btmorr