Opentrons/opentrons #2346: Manage Robot Connection: Updated Robot Connection Card

umbhau, 2018-09-23:
As a Run App user, I would like to be able to see full details of my robot's connection state, so that I can make informed decision when managing my robot's connection.

## Acceptance criteria
- Update 'Robot Connection' card information as follows:
  - Display active connection method
    - "Your robot is currently connected via [ Wifi | USB ]
  - Display USB connection details
    - Wired IP Address (is 'wired' the most accurate/intuitive term?)
    - Wired MAC Address 
  - Display WiFi connection details and management options
    - Wireless IP Address
    - Wireless MAC address
    - Dropdown filed to select/add WiFi connections (functionality not covered in this user story)

## Design
Following sketch is just for discussion/estimation purposes. Final design tbd from @pantslakz.

umbhau, 2018-09-26:
@mcous @pantslakz re: above open question, I'm suggesting we can hold of on persistently displaying connection status for both USB and WiFi, but should include internet status.

Assuming internet status is just 'Connected to Internet' or 'Not Connected to Internet', but please correct me on any of this.

umbhau, 2018-09-26:
Agreed on the following with @mcous:

- Should we include a visible 'connection status' for both Wifi and USB, regardless of whether it is the actively used connection method? E.g for USB: 'Disconnected' or 'Connected'. This is obviously a little tricky with the word 'connectivity'. **_No_**
- Should we further include a visible differentiation of 'actively used' vs 'inactive' connection methods? 
- E.g. for USB: 'Disconnected', 'Connected - Inactive', or 'Connective - Active' - No? **_No_**
Should we include internet status in this card? If so, what are the options? - Yes? "Connected" | "Not Connected" **_Yes_**

pantslakz, 2018-09-27:
To keep copy short, suggest "Your app is currently connected to your robot via [ Wifi | USB ]"

commit 0af61f06 (2018-10-15): refactor(app): Add new ConnectionCard with interface and WiFi details

Closes #2346, closes #2347
PR #2484 (2018-10-16): refactor(app): Add new ConnectionCard with interface and WiFi details
## overview

This PR closes #2346 and closes #2347 by adding a new connection card to the robot settings page that displays the robot's:

- Internet connectivity status
- Current connection to the app (USB or Wi-Fi)
- IP and MAC address of the Wi-Fi interface
- IP and MAC address of the Ethernet interface
- Available Wi-Fi networks with each network item displaying:
    - A checkmark if the network is currently active
    - A lock if the network is secured
    - A signal strength indicator

Note: Wi-Fi select list will requi [... cut]
closed 2018-10-16 by mcous