Opentrons/opentrons #2354: Robot update: communicate upgrades and downgrades more clearly

mcous, 2018-09-24:
## overview

As a user, I would like my app to warn me if my robot's software is out of sync with my app so that I may take some action depending on my needs (see below).

A user may want to:

- Always be running the latest version of everything
- Stay on one specific version without updating
- Switch back and forth between beta and stable versions

Regardless of whether or not the app is fully up to date, if the user wishes to use that specific version of the app, the robot must be on a matching release to guarantee compatibility.

## current behavior

The app will alert the user if there is a version mismatch between the app and the robot, but this alert is generic and always uses language that suggests the user is upgrading the robot, when they may, in fact, be downgrading the robot.

## acceptance criteria

There are 4 scenarios in which a version mismatch could happen, and the app should message to the user accordingly:

- [ ] The app cannot find an update and its version is greater than the API
    - App should prompt the user to either "upgrade" their robot (default)  or
    - Ignore the mismatch
- [ ] The app cannot find an update and its version is lower than the API
    - App should prompt the user to "downgrade" their robot (default) or
    - Ignore the mismatch
    - Message should tell the user to ensure their computer is online so the app can ensure it is fully up to date
    - Example 1: Robot is on a beta version, app is on stable, needs to be downgraded to stable
    - Example 2: Robot was updated to latest stable by another user, current user is running an older app but is currently not of wifi, so user 2's app doesn't know it's out of date
- [ ] The app _can_ find an update _and_ its version is greater than the robot's
    - The app should prompt the user to go to the App info page to update their app (default) or
    - Upgrade their robot to the out-of-date version or
    - Ignore the mismatch
- [ ] The app _can_ find an update _and_ its version is lower than the robot's
    - The app should prompt the user to go to the App info page to update their app (default) or
    - Downgrade their robot to the out-of-date version or
    - Ignore the mismatch

mcous, 2018-10-01:
Re-opening pending final polish after design pass. Skeleton functionality / copy is in, though

Kadee80, 2018-10-02:
Closing with current functionality/copy in favor of opening a separate design/UI pass ticket.

commit cd3971f1 (2018-09-27): feat(app): Communicate upgrades and downgrades more clearly

closes #2354
PR #2386 (2018-09-28): feat(app): Prompt user to update app in robot update modal
## overview

This PR closes #2354 by prompting user to update thier app in the update robot modal if there is an app update available.

## changelog

- feat(app): Prompt user to update app in robot update modal

## review requests

### app and robot are equal
- [ ] robot update buttons  = [reinstall]
- [ ] update robot modal displays update app message

### app is behind robot version
change app and app-shell package.json version to 3.3.0
- [ ] update robot modal displays update app message

### app is ahead of robot
change app and app [... cut]
commit c3897503 (2018-09-28): feat(app): Prompt user to update app in robot update modal (#2386)

closes #2354
commit 644944a8 (2018-11-08): Update PD Staging Beta (#2646)

* refactor(app): Add state for API update file (#2364)

* feat(api): Add skeleton of new Labware class (#2363)

Closes #2261

* fix(api): Change api update ignore route to be accessible to client (#2368)

Closes #2367

* feat(api): change tempdeck temperature resolution,fix temp in docs (#2359)

Closes #2358

* feat(app): Add upgrade and downgrade logic to robot updates (#2376)

* chore(api): Add 'TODO's to labware-related tests (#2372)

* chore(api): Add 'TODO's to labware-related tests. Closes #2259

* chore(api): Fix some mypy typing errors (#2387)

* feat(ap [... cut]
closed 2018-10-02 by Kadee80