Opentrons/opentrons #2365: API regression: /server/update/ignore no longer functions

mcous, 2018-09-26:
## overview

We added `GET+POST /server/update/ignore` in #1693 so that a robot could tell the app that it was ignoring a certain update (so that the app would not pop up an update alert modal). #1701 subsequently broke that functionality. Out of an abundance of caution, if the app cannot hit `/server/update/ignore`, it does not pop up the alert modal.

As we improve the robot update alert modal in the app, we need this functionality back so that the user is actually alerted to updates

## current behavior

`POST /server/update/ignore` 404s

## steps to reproduce

1. Open the app with devtools on
2. Click on a robot
3. Note the failure to hit `/server/update/ignore`

## expected behavior

`/server/update/ignore` should function like it did when it was added in #1693

sfoster1, 2018-09-26:
Does this also happen with `server/update/firmware`?

mcous, 2018-09-26:
Duplicated and closed by #2367

closed 2018-09-26 by mcous