Opentrons/opentrons #2381: Feature Request: track tip rack state as a property of the tip rack labware rather than the pipette.

emernic, 2018-09-26:
## overview

Changing tip tracking to be a property of the labware rather than pipettes would simplify complex use cases such as tip sharing between compatible pipettes, implementing custom tip iteration for applications such as cherry picking using multichannels, and serializing tip state so that LIMS systems can "remember" which tips have been picked from which tip boxes.

## current behavior
Current state of which tips have been used or not is tracked using an iterator for each pipette. Tip state cannot be determined by inspecting the tip rack labware item object. (if I am misunderstanding please let me know).

## desired behavior

Tip rack objects have some property such as `my_tiprack.unused_tips` which returns an array of the wells which still have tips in them. This array would be updated whenever tips are picked.

mcous, 2019-08-28:
Related: #2176

sfoster1, 2019-12-13:
This is how it works in Protocol API V2!

closed 2019-12-13 by sfoster1