Opentrons/opentrons #2388: OT2 can't ping connected PC via USB/Eth

simonchih, 2018-09-28:
## overview
Local PC connected OT2 using USB/Eth (IP:169.254.x.x). Then Local PC connect OT2 via SSH. I found user can't get response by ping local PC IPv4 in SSH Terminal (OT2 side), but local PC can ping OT2's IPv4. Is it defect? or just work as design? 

## current behavior
User can't get response by ping local PC IPv4 in SSH Terminal (OT2 side), but local PC can ping OT2's IPv4. It seems local PC can ping OT2 but can't in reverse way.

## steps to reproduce
precondition: Local PC connected OT2 using USB/Eth, e.g. Local PC IP:169.254.101.24, OT2 IP:169.254.91.23.
1. Local PC ping OT2's IPv4 is successful, e.g. 169.254.91.23.
2. Local PC connect to OT2 via SSH
3. In SSH terminal, ping local PC IPv4 (e.g. 169.254.101.24) fail
==> Can't get any response by ping

## expected behavior
In SSH terminal (OT2 side), ping local PC IPv4 (e.g. 169.254.101.24) is successful.

mcous, 2018-09-28:
Hi, thanks for the report! I can confirm there is nothing about the OT2's configuration that would prevent outbound pings. I just SSH'd into a robot via the USB-to-Ethernet connection and was able to ping my own computer without issue.

The most likely cause of this is that your local PC is configured to not respond to ping (ICMP echo) requests on that network interface. I would suggest double checking any firewall or security settings on your computer

closed 2018-09-28 by mcous

simonchih, 2018-10-01:
Yes, you are right. It's blocked by firewall.