Opentrons/opentrons #2393: PD: "not enough tips" error not showing in some distributes

IanLondon, 2018-10-01:
## overview / current behavior

When a distribute is not the first step (?) and uses more than the number of tips available, the "not enough tips" error is correctly generated (in substeps), but doesn't show up as a visual error in the timeline.

Maybe it's not being generated in the timeline, but only in substeps??

Reported by 2 beta users

Reproduced using PD dev, commit 0202b535e609501a5b6f66a99c36928f74cb9a4f

## steps to reproduce

- Use up some tips in previous step(s)
- Make a distribute that uses more than the number of tips available

## expected behavior

"Not enough tips" error should show. It's being generated (visible in console log "Could not get substep, had errors"), but not shown like a normal error.

IanLondon, 2018-10-24:
Cannot reproduce in c6a9f38b3468eac902d1707fb9b14d94e3d115ad

umbhau, 2018-10-25:
@howisthisnamenottakenyet

pantslakz, 2018-12-06:
@IanLondon is this still happening?

IanLondon, 2018-12-06:
Seems like this isn't happening, closing

closed 2018-12-06 by IanLondon