Opentrons/opentrons #2453: PD: Add support for 'opentrons-tuberack-15_50ml'

IanLondon, 2018-10-10:
## overview

As a PD user, I want to be able to use the new Opentrons 4-in-1 rack with the 15x50 tube rack insert.  Also, I want to be able to easily tell apart the 2 15x50 tube racks (old acrylic "DIY" vs the new 4-in-1) by their names in the labware selection modal.

## acceptance criteria
- [ ] In labware selection modal, rename the existing 15x50 to "15mL x 6 + 50mL x 4 Tube Rack (Clear Acrylic)"
- [ ] Add a new option for "Opentrons 15mL x 6 + 50mL x 4 Tube Rack (4-in-1 Rack)" and give it the image below. This is for labware def `opentrons-tuberack-15_50ml`

## image
[image]

howisthisnamenottakenyet, 2018-10-10:
What about
"15mL x 6 + 50mL x 4 Tube Rack (Opentrons clear)" 
"15mL x 6 + 50mL x 4 Tube Rack (Opentrons black)"
or something more to that effect. Or we could do short/tall like with the well plates. I wouldn't call one Opentrons and one not because they're both our tube racks so that could get confusing.

@pantslakz @umbhau

umbhau, 2018-10-10:
I'd keep the name for the new labware simple + in line with our other defs, since it effectively deprecates the old tube rack.

umbhau, 2018-10-10:
FYI this is the naming convention laura is proposing for new labware: [brand]_[num.wells]_[displayCategory]_[totalVol]_[displayVolumeUnits]_[OTuniqueID]

IanLondon, 2018-10-10:
@umbhau are we dropping support for creating new protocols in PD with the old tube rack? I thought some people are still using it, but maybe there aren't that many of them who will use PD? Idk.

Laura's proposed naming convention is for the .py protocol world only, that generated name is a "load name" and is effectively a programmer-readable ID, used only to call `labware.load(slot, loadName)` in .py protocols. 

PD cannot see the "load name" and doesn't use it, and we don't have plans to use it in PD -- it's solely a concern of .py protocols

We don't have a convention for the human-readable `displayName` which is what PD users would see.

PR #2530 (2018-10-25): feat(protocol-designer): add 4-in-1 15-50 tuberack to PD
## overview

Closes #2453

[image]

## changelog

* rename existing tuberack
* add 'opentrons-tuberack-15_50ml'
* make labware selection modal wider, to accommodate longer names
* move labware diagram images into `protocol-designer/src/images` instead of using OT2 docs image URLs. (These get bundled as base64 data URLs via `url-loader`, like the other images in PD)

## review requests

- [ ] Labware hover diagrams still show up (in `make -C protocol-designer/ dev`)
- [ ] Also show up on the branch build: http://opentrons-protocol-designe [... cut]
commit 48cc9c02 (2018-10-25): feat(protocol-designer): add 4-in-1 15-50 tuberack to PD (#2530)

* rename existing tuberack
* add 'opentrons-tuberack-15_50ml'
* include labware diagram images in protocol-designer src and include in bundle.js as data URLs, instead of just linking to docs.opentrons.com/_images/

Closes #2453
closed 2018-10-25 by IanLondon