Opentrons/opentrons #2509: bug(api): Delay does not work for 1 hour + actions

Laura-Danielle, 2018-10-17:
## overview

As a user, I would like the delay function to work for up to 12-15 hours.

## current behavior
Delay only works for times less than 1 hour. Otherwise, it ends the protocol session and clears any memory of the current protocol being run.

## steps to reproduce

1. Write a protocol using pipette.delay(minutes=60)
2. Observe robot completing the delay, but then not finishing out the rest of the protocol
3. Check the logs to see that the session has been interrupted

## expected behavior

Protocol should not fail if delay is longer than 1 hour

btmorr, 2019-05-06:
I think this has been fixed, but I can't find the reference to the work. @Laura-Danielle @sfoster1 @umbhau do you have info on any outstanding issues, or can this be closed?

mcous, 2019-08-29:
Confirmed with @Laura-Danielle that this is no longer an issue

Test results (internal Opentrons doc): https://docs.google.com/document/d/19LFrwrRwzgDVdvLcrL-jSS_6kuIFiz4uHBHoPGFqB_Y/edit?pli=1

closed 2019-08-29 by mcous