Opentrons/opentrons #2567: Install the App for all users

cmfield, 2018-10-26:
Is it possible to install the App on Windows for all users of the computer?

mcous, 2018-10-26:
Hi @cmfield, thanks for the request! By default, the app is configured to both:

1. Not require an install wizard (i.e. use a one-click installer)
2. Install the app per-user (instead of per-machine)

Per-machine isn't an option for our users in environments where they don't have IT permission to install applications system-wide, so if we allowed (2) to be configurable we'd also have to switch (1) from a one-click installer to an installation wizard.

I'll ask our UX team about this, but in the mean time: for you personally, would the annoyance of having to click more buttons to install the app be worth it to gain the ability to install per-machine?

mcous, 2018-10-26:
Unfortunately it's a [build time configuration](https://www.electron.build/configuration/nsis) for us, so we'd have to change it for all Windows users. We'll run through it on our end and see what it looks like; given this is something you need we'll likely end up enabling configuration in the wizard. I'll keep this thread updated

commit 4f6c0fb7 (2019-09-06): fix(app-shell): Update Electron and add macOS app notarization

Closes #3997, closes #2567
commit 6b87876d (2019-09-06): fix(app-shell): Update Electron and add macOS app notarization

Closes #3997, closes #2567
commit 91c21d4e (2019-09-06): fix(app-shell): Update Electron and add macOS app notarization

Closes #3997, closes #2567
PR #4011 (2019-09-10): fix(app-shell): Update Electron and add macOS app notarization
## overview

This PR updates a lot of our underlying Electron / app packaging tooling to ensure we are ready for new security requirements in the upcoming version of macOS (10.15 Catalina).

This Electron update will bring reliability and security improvements to all our users, but out of necessity, is going to introduce a little bit of user pain for our Linux and Windows users out of the gate in exchange for a better long term experience.

Closes #3997, closes #2567 

### the good

- When macOS 10.15 is launched in the coming week [... cut]
closed 2019-09-10 by mcous