Opentrons/opentrons #2587: Protocol Designer: Clean up Selected Container Getter

b-cooper, 2018-10-31:
getSelectedContainer & getSelectedContainerId are almost the same, probably go for getSelectedContainerId (and a few other redundant selectors), rename as Labware?

Let's rename to `getSelectedLabware` and `getSelectedLabwareId` and determine where we can just use the id where, we are currently grabbing the whole entity.

IanLondon, 2018-12-06:
@b-cooper are you OK to close this?

I think we did the important parts (?), the remaining thing in the ticket is that`getSelectedLabware` and `getSelectedLabwareId` could be merged into one.

I'd prefer they weren't merged because 1) `getSelectedLabware` returns `?Labware` and that Maybe lets Flow keep us from getting a set of whitescreening errors we used to get when deleting labware and 2) if we only have `getSelectedLabwareId`, we have to more verbosely do `const allLabwareById = selectors.getLabwareById(state); const myLabware: ?Labware = allLabwareById[selectors.getSelectedLabwareId(state)]` in 6 places which is more redundant than having those 2 selectors and using whichever is more appropriate

b-cooper, 2018-12-06:
@IanLondon Yeah that sounds reasonable.  Go ahead!

closed 2018-12-06 by IanLondon