Opentrons/opentrons #2599: Marketing Analytics: LinkedIn Insight Tag

umbhau, 2018-11-01:
As an Opentrons marketer, I would like to have the LinkedIn Insight tag on our website, so that I can retarget our website visitors on LinkedIn.

## Acceptance Criteria
- [ ] Add Insight tag to opentrons.com

## Open Questions
- Full JS insight tag or just [image pixel?](https://www.linkedin.com/help/lms/answer/65521/the-linkedin-insight-tag-frequently-asked-questions?lang=en)
- Just opentrons.com, or other properties as well?

## Tag
[add tag here]

mmonsees, 2018-11-01:
The Insight tag should be installed on subdomains www/protocols/shop as outlined in their specs. This can be done first if necessary, as it is both easier and more essential to building up our audiences. 

[linkedin insights tag.txt](https://github.com/Opentrons/opentrons/files/2540160/linkedin.insights.tag.txt)

The second part of this is to measure conversions using pixel tracking since we don't have confirmation pages that would otherwise trigger rules-based conversions. The pixel tracking process, which is outlined [here ](https://www.linkedin.com/help/linkedin/answer/93958), looks like it is most easily implemented via GTM though I'm not sure we're using that for our sites currently.

[linkedin conversion schedule demo.txt](https://github.com/Opentrons/opentrons/files/2540161/linkedin.conversion.schedule.demo.txt)

[linkedin conversion email team.txt](https://github.com/Opentrons/opentrons/files/2540169/linkedin.conversion.email.team.txt)

[linkedin conversion chat team.txt](https://github.com/Opentrons/opentrons/files/2540171/linkedin.conversion.chat.team.txt)

closed 2018-11-29 by umbhau

mmonsees, 2018-12-14:
@umbhau  so I understand, why would a ticket like this be closed?

umbhau, 2018-12-14:
hey @mmonsees -- my understanding was that you and Mike decided to tackle these tags through GTM.