Opentrons/opentrons #2601: Marketing Analytics: Verify Google Merchant Center via Shopify

umbhau, 2018-11-01:
As a Opentrons marketer, I would like to verify shop.opentrons.com in the Google Merchant Center.

## Acceptance Criteria
- [ ] Upload verification file (?) to shop.opentrons.com

## Implementation Notes
From Matt: 
```Upload the HTML file to your website (most common)

To use this method, you must be able to upload new files to your server.
1. Click the link to download this HTML file:googlead3ca12fcb6b12fc.html
2. Upload the file to https://shop.opentrons.com
3. Confirm the upload by visiting https://shop.opentrons.com/googlead3ca12fcb6b12fc.html.
4. Click Verify URL to submit your URL to Google Search Console for verification and claim it.
Once your submit your URL, you'll be sent a confirmation email from Google Search Console.
To stay verified, don't remove the HTML file, even after your URL is verified.

File: https://drive.google.com/open?id=1Hv18KdSVwuAKnvPGbzhKg-kv2r3EOLYp (same as above I think)```

mmonsees, 2018-11-15:
@Kadee80 @umbhau Hey, this one is probably a 20 minute project tops. Could I ask that you take a look and add the snippet if it looks as simple as I think it is? It is holding me back from an otherwise simple program setup..

Kadee80, 2018-11-15:
@mmonsees I can take a look before end of week

mcous, 2018-11-15:
Done via DNS instead. opentrons.com and all subdomains are now Google verified

closed 2018-11-15 by mcous