Opentrons/opentrons #2636: Unique labware name does not refer to unique calibration data

vanessa-king, 2018-11-06:
## overview
Bug in the function of the 'unique name' third optional argument in labware.load().

## current behavior
When trying to place multiple identical containers in the same slot using the share=True feature, adding a unique name as the third optional argument has no effect on the calibration data used. Currently in the API, it says: 
" A third optional argument can be used to give a labware a unique name.

p = labware.load('96-flat', '2', 'any-name-you-want')

Unique names are useful in a few scenarios. First, they allow the container to have independent calibration data from other containers in the same slot. In the example above, the container named ‘any-name-you-want’ will assume different calibration data from the unnamed plate, even though they are the same type and in the same slot."

This does not happen in practice, but rather every container of the same type uses the same calibration data, no matter the unique name. 

## steps to reproduce
Use the API example: 

fa = labware.load('T25-flask', '3', 'flask_a')
fb = labware.load('T25-flask', '3', 'flask_b', share=True)
fc = labware.load('T25-flask', '3', 'flask_c', share=True)

The App will only ask for the calibration of a 'T25-flask' container once.

## expected behavior
Labware given a unique name should use its own calibration data. The App should list the uniquely-named containers as separate containers and go through the calibration process for each one.

mcous, 2018-11-08:
Hey @vanessa-king, that was my bad! I thought I responded to you but turns out I forgot to hit "Comment" because my brain wasn't working.

Unfortunately, it looks like this part of our documentation is completely wrong, we do not use the `label` parameter to separate out calibration data, and I don't think we have any plans to at this stage. That's why I added a "docs" to this ticket.

Please correct me if I'm wrong, but I believe that this particular use case you're referring to in your request is:

1. Place a number of free-standing flasks arbitrarily into the same slot
2. Use the calibration procedure to "trick" the robot into knowing where each of those flasks are in that slot
3. Proceed with the protocol using those flasks in those positions

Given how the system is designed, I use the word "trick" above because this is technically a misuse of the calibration system that _never_ should've been appropriated for this purpose, much less recommended to users. Accordingly, I'm working on two things right now:

1. Getting you a workaround that I will post here so you can do your science stuff!
    - It will likely involve using `labware.create` to build three separate "custom" flasks
    - If you had some sort of fixture to hold the flasks in place at known distances, this would be a single `labware.create` call
    - It should look fairly similar to what the docs currently say, using `share=True`, etc.
2. Talking to our science and UX teams to learn about this particular use case so the software team can build the appropriate solution
    - This is what the "api" label I added to the ticket is for

Sorry for the delay on this, and I'll ping you again here when we have that workaround protocol for you.

vanessa-king, 2018-11-08:
Hi @mcous, thank you for the reply. Don't worry, I do have an actual fixture that I am using! The wells in my container do not have consistent `spacing` between columns. It was recommended to me through Opentrons support to instead define my container as multiple containers that do have consistent `spacing`. This way, I can describe my 1 container as 4 containers in one slot.

Using this "multiple containers" method to describe my container, I used `share=True` to place the multiple containers in the same slot. But as you said, I found the `label` parameter does not separate out calibration data.

Certainly a valid workaround for this is to define all 4 sub-containers as different containers so that they can have their own calibration data. As a user, I was hoping there could be a neater way to do this rather than creating multiples of the same container? Whether that be by allowing the `label` parameter to refer to different calibration data or by making the `spacing` parameter in `labware.create()` more flexible.  If this isn't possible in the foreseeable future, that is fine as I do have a valid workaround.

mcous, 2018-11-08:
> The wells in my container do not have consistent spacing between columns

Oh, I see! That makes a lot of sense, and I can see a potential path forward on getting `labware.create` to support this kind of use-case. I will dig into that a little and keep you posted

closed 2022-03-04 by mcous