Opentrons/opentrons #2641: Bug: Fail safe when tip ejection is not successful

straussmaximilian, 2018-11-08:
## overview

When tip ejection is not successful the robot continues protocol and rams tip into rack when trying to pickup new tip. This leads the pipet to be clamped and the tip rack is dragged along.

## current behavior

Currently, the protocol continues, if not stopped manually in time the material area can be damaged. 
## steps to reproduce
Probably make a calibration with a very tight seal.

## expected behavior

The protocol should stop. 
Is there any way to monitor the force exerted by the robot and set an upper limit?
Alternatively adjust calibration guidelines to avoid too tight calibration of tip rack.

straussmaximilian, 2018-11-09:
Hi @mcous, thanks for your reply.

- The pipette is a P10.
- I had 3.40 and edge6168d32
- No motor configuration changes were made

Today I did the following, and the problem persisted (I attached a picture for your reference):
- Update and reset of the robot
- Calibration of pipette and labware -> Error after ~ 25 tips
- Calibration of pipette and labware, I tried to be a bit above being flush, as recommended (picture attached) -> Error after ~ 5 tips
- Calibration (the one with the crosses), Calibration of pipette and labware, I tried to be a bit above being flush, as recommended (picture attached) -> Error after ~ 5 tips

From my observations, it looks as if the robot is using way to much force when picking up the tips.

Hope that helps.
[image]
[image]

mcous, 2018-11-13:
@straussmaximilian I talked with our hardware team, and we think this is most likely an issue with the motor current settings. We've got in-progress work to refine these settings as well as tighten the QA/QC process for the hardware itself. Please contact our support team via https://support.opentrons.com or using your in-app support chat for temporary workarounds until our engineering work is completed.

Since the process itself (jog to top of tip, pick it up, verify) is "correct" and we don't have the ability to detect failed drop tip, I'm going to close this issue in favor of existing in-flight work to get these current values correct.

closed 2018-11-13 by mcous

mcous, 2018-11-13:
This _may_ be a result of a bug where drop-tip currents weren't being properly applied. This issue was fixed in #2572 which will be included in the 3.6.0 release