Opentrons/opentrons #2651: Disconnect from robot not disconnecting

straussmaximilian, 2018-11-09:
## overview
When toggling the disconnect switch in the Robots list, the robot seems still be connected.

## current behavior
Toggling the switch apparently connects and disconnects. Upon connecting, a green message displays the successful connection. However, when being disconnected, the lights can still be switched on and off.

## steps to reproduce
Disconnect robot and toggle Lights switch.

## expected behavior

After disconnection one should not be able to toggle switches.

mcous, 2018-11-12:
Hi @straussmaximilian, thanks for the report and sorry for the confusion here! The app uses two different interfaces (or "APIs") to talk to the robot:

- Real-time API for running protocols
    - Connected when you click the "Connect" toggle or button
- HTTP API for doing everything else
    - Ephemeral connections made per-request

The lights use the HTTP API, so they do not require the real-time API to be connected to function.

closed 2018-11-12 by mcous

mcous, 2018-11-13:
This _may_ be a result of a bug where drop-tip currents weren't being properly applied. This issue was fixed in #2572 which will be included in the 3.6.0 release