Opentrons/opentrons #2652: Feature flag to turn PD hints back on

howisthisnamenottakenyet, 2018-11-09:
As a user I'd like to be able to restore my hints modals

## Acceptance criteria
- [ ] Add a new sidebar nav item to Settings tab
- [ ] Have a method for restoring hints

howisthisnamenottakenyet, 2018-11-13:
@umbhau @pantslakz is this good to go?

umbhau, 2018-11-13:
@howisthisnamenottakenyet looks good to me. could see us eventually just wanting an 'advanced settings' section like RA, and moving this in there...but with only one thing, this seems :+1::skin-tone-4:

pantslakz, 2018-11-14:
> @howisthisnamenottakenyet looks good to me. could see us eventually just wanting an 'advanced settings' section like RA, and moving this in there...but with only one thing, this seems 👍:skin-tone-4:

Looks good, would just update copy:
“Restore all hints and tips modals”

howisthisnamenottakenyet, 2018-11-14:
@pantslakz I believe modal is a dev/design term, and we can't expect everyone in our user base to know what it means.

pantslakz, 2018-11-20:
True enough - how about "Restore all hints and tips notifications"

howisthisnamenottakenyet, 2018-11-20:
@pantslakz updated

PR #2726 (2018-11-29): feat(protocol-designer): allow user to re-enable dismissed hints
## overview

Closes #2652

## changelog

* add action + reducer + button to clear dismissed hints state
* rename the sidebar itiem from "Privacy" -> "App", and correspondingly rename the `Privacy` components to `SettingsApp`

## review requests

* Click the Restore button
* Make a new protocol and add a Transfer without adding liquids, you should see a hint
* Click OK to dismiss the hint (you can check "Don't show me again", or not, either way it will get restored when you click the button)
* Clicking the Restore button will un-d [... cut]
commit af52d1e1 (2018-11-29): feat(protocol-designer): allow user to re-enable dismissed hints (#2726)

Closes #2652
closed 2018-11-29 by IanLondon