Opentrons/opentrons #2688: Optional mix arguments in transfer not compatible with .bottom() or .top()

emernic, 2018-11-16:
## overview / current behavior

Calling `my_pipette.transfer(10, source, dest.bottom(), mix_after=(2, 10))` causes the mix step to happen in the center of the well instead of the bottom. This is a little surprising and also can cause the mix to fail if (for example) the user only has a small volume in the well.

This happens because the `_dispense_during_transfer` function [here](https://github.com/Opentrons/opentrons/blob/d7c7f6066f4ab4aaa2c03c583bbe0d4568187bba/api/src/opentrons/legacy_api/instruments/pipette.py) only passes the well (not the actual location) to `_mix_during_transfer`.

## expected behavior
Mix step should probably happen at the location specified. The only use case I can think of in which this would be undesirable is if the user were trying to do a contact-free transfer (and therefore set the destination to `my_well.top()`), however, this is useless if you are mixing anyway (because the tip has to tough the liquid).

Second best option would be to at least document the argument's behavior clearly in the user docs (regardless of which way it works).

bndo, 2019-09-03:
Need to verify if this will be fixed with the launch of V2

mcous, 2021-12-10:
Closing as stale given PAPIv1 was removed some time ago

closed 2021-12-10 by mcous