Opentrons/opentrons #2694: Execute Python script via SSH console, show OSError: [Errno 30] Read-only file system

simonchih, 2018-11-19:
## overview
After calibration, we execute python script on OT2's SSH console, it runs fine but show "OSError: [Errno 30] Read-only file system", why?

## current behavior
we execute python script on OT2's SSH console, it runs fine but show error message below:
```
5c8b8de:~# python thread_tcp_client_OT2.py
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
```

## steps to reproduce
1. Ope Opentrons OT2 app
2. Connect to OT2
3. Using [thread_tcp_client_OT2.py](https://drive.google.com/open?id=1O2TyFo12i-tMPJKjbCjRocaRJD0EIK-W), remark robot.connect() and socket related function as comment to do calibration.
4. Don't run script by the app after calibration but close the GUI app
5. Connect OT2 port 22 (SSH) by PuTTY
6. Login as: root
7. Put thread_tcp_client_OT2.py on TFTP server
8. Download thread_tcp_client_OT2.py to OT2's root directory:
:~#  tftp -g -r thread_tcp_client_OT2.py <TFTP Server's IP>
9. Run server side python script (can't provide now)
10. In OT2's SSH console, execute thread_tcp_client_OT2.py:
python thread_tcp_client_OT2.py
==>
 It will show error message below, and then script works fine.
```
Unable to move settings data due to:
Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/config/__init__.py", line 117, in _move_settings_data
    shutil.copy2(pth, dest_path_dict[key])
  File "/usr/local/lib/python3.6/shutil.py", line 257, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/local/lib/python3.6/shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
OSError: [Errno 30] Read-only file system: '/mnt/usbdrive/config/pipetteData.json'
...
```

Note: You can't run the thread_tcp_client_OT2.py due to server side's script is NOT provided.

## expected behavior
In OT2 version v3.3.0,
there is NO any error  message with same python script run at SSH console.
I want to know how to work without error message?
If there is nothing to do, I expect there is NO error message at SSH console.

btmorr, 2018-11-19:
This happens if the OS fails to mount the attached thumb drive correctly. Unfortunately the only fix I've been able to find so far is to unplug the thumb drive and plug it back in. It can be located by removing the cover in the upper-right-back corner of the inside of the robot (the screws are on the back of the robot).

simonchih, 2018-11-20:
I don't know where is thumb drive? Could you take  a picture?

Laura-Danielle, 2018-11-26:
Hi @simonchih see the support article [here](https://support.opentrons.com/ot-2/troubleshooting/unexpected-error-saving-deck-calibration).