Opentrons/opentrons #2695: When a PD user switches tip rack size for a pipette, all tip racks belonging to that pipette should switch to that size

howisthisnamenottakenyet, 2018-11-19:
As a user I'd like to not click unnecessarily. A user who switches a tip rack type for a pipette has to go to deck setup and manually replace all the old tip racks with the new tip racks. 

## Acceptance Criteria
- [ ] When a user switches  tip rack size for a pipette, all tip racks belonging to that pipette should switch to that new size

b-cooper, 2018-12-05:
Blocked by #2753

b-cooper, 2018-12-10:
Putting this on the icebox for now as any close at hand implementation puts us in a confusing middle state with the UI. Let's revisit this when we can make a completely educated transformation to the tipracks on pipette switch.

With sharing tipracks, the ideal behavior here would be to delete all unnecessary tipracks and add exactly the number of tipracks required for all of the current steps. In order to do this we need to construct a selector that returns an object mapping of pipetteId to a map keyed by tip type (tiprackModel) of number of tips used by the final deck state.

 This selector could also be very useful down the road for smart suggestions about tip usage and possibly a split between, sterile mode: no human intervention steps / aim for economy of space on the deck, and efficiency mode: as many human interventions steps as needed (within reason). This efficiency mode would include the new features. It would allow you to place imperative labware whose location should be fixed and respected. Then it could take the unused slots and suggest tips/tipracks to fill those slots with in order to achieve the goal in the shortest amount of time.  This could extend to pipette selection and tip usage efficiency as well.

closed 2022-09-30 by bndo