Opentrons/opentrons #2718: Disconnect from Wifi

pantslakz, 2018-11-27:
As a robot user, I would like to be able to disconnect from a wifi network.

## Acceptance Criteria
-  [ ] User able to view "Disconnect from (name of network)" when connected to network in wifi dropdown
-  [ ] If user is not connected via USB when attempting to disconnect, "Disconnect from (name of network)" disabled, with tooltip letting them know they must connect via USB to disconnect/change network
-  [ ] Upon selecting disconnect, user asked to confirm

## Design
- https://zpl.io/anY8n9Y
- https://zpl.io/beLx9vl
- https://zpl.io/bA43g4O

## Outstanding questions
- should we also disable other network names, unless user is connected via USB?

pantslakz, 2018-11-27:
@mcous view here for button width: https://zpl.io/bAqJyXp

mcous, 2020-01-24:
Superseded by #4817

closed 2020-01-24 by mcous