Opentrons/opentrons #2748: "Get new tip: only the first aspirate" not functioning correctly

pantslakz, 2018-11-30:
User set "Get new tip: only the first aspirate" in dispense, however it picks up a tip for every aspirate.

pantslakz, 2018-11-30:
`{
    "protocol-schema": "1.0.0",
    "metadata": {
        "protocol-name": "CuvetteFilling1",
        "author": "",
        "description": "",
        "created": 1543481814071,
        "last-modified": 1543483971437,
        "category": null,
        "subcategory": null,
        "tags": []
    },
    "default-values": {
        "aspirate-flow-rate": {
            "p10_single": 5,
            "p10_multi": 5,
            "p50_single": 25,
            "p50_multi": 25,
            "p300_single": 150,
            "p300_multi": 150,
            "p1000_single": 500
        },
        "dispense-flow-rate": {
            "p10_single": 10,
            "p10_multi": 10,
            "p50_single": 50,
            "p50_multi": 50,
            "p300_single": 300,
            "p300_multi": 300,
            "p1000_single": 1000
        },
        "aspirate-mm-from-bottom": 1,
        "dispense-mm-from-bottom": 0.5,
        "touch-tip-mm-from-top": -1
    },
    "designer-application": {
        "application-name": "opentrons/protocol-designer",
        "application-version": "v3.6.0-beta.0-2-g941916fb",
        "_internalAppBuildDate": "Wed, 14 Nov 2018 18:39:35 GMT",
        "data": {
            "pipetteTiprackAssignments": {
                "pipette:left:p300_single": "opentrons-tiprack-300ul",
                "pipette:right:p50_single": "opentrons-tiprack-300ul"
            },
            "dismissedWarnings": {
                "form": {},
                "timeline": {}
            },
            "ingredients": {
                "0": {
                    "name": "MilliQ",
                    "description": "MilliQ Water",
                    "serialize": false,
                    "liquidGroupId": "0"
                },
                "1": {
                    "name": "BC",
                    "description": "Berberine Chloride BC-2018-07-23",
                    "serialize": true,
                    "liquidGroupId": "1"
                },
                "2": {
                    "name": "CB8",
                    "description": "CB8 LauraA",
                    "serialize": true,
                    "liquidGroupId": "2"
                }
            },
            "ingredLocations": {
                "04908390-f3b5-11e8-b9a8-731205eb7eec:tube-rack-2ml": {
[... 1066 lines cut ...]
                        "labware": "f2ea14d0-f3b4-11e8-b9a8-731205eb7eec:24-well-plate",
                        "well": "C6",
                        "offsetFromBottomMm": 0.5
                    }
                },
                {
                    "command": "blowout",
                    "params": {
                        "pipette": "pipette:right:p50_single",
                        "labware": "trashId",
                        "well": "A1"
                    }
                },
                {
                    "command": "drop-tip",
                    "params": {
                        "pipette": "pipette:right:p50_single",
                        "labware": "trashId",
                        "well": "A1"
                    }
                }
            ]
        }
    ]
}`

IanLondon, 2018-12-05:
temporarily blocked by #2773 (touching a lot of step-generation)

IanLondon, 2018-12-06:
Reported protocol as example: [CuvetteFilling1.json.zip](https://github.com/Opentrons/opentrons/files/2653545/CuvetteFilling1.json.zip)

This is happening in Distribute when the distribute volume exceeds the capacity of the pipette/tip, so it gets broken into multiple transfers. Each inner transfer was getting the same 'changeTip="once"', so it was getting a new tip every inner transfer "chunk".

PR #2784 (2018-12-06): fix(protocol-designer): fix changeTip once bug in distribute step
## overview

Closes #2748 - see my comment in that ticket for explanation of problem

## changelog

## review requests

- When Volume of a Distribute exceeds the pipette/tip capacity, change tip "once" should use one tip throughout the entire step
commit 64111f65 (2018-12-06): fix(protocol-designer): fix changeTip once bug in distribute step (#2784)

Closes #2748
closed 2018-12-06 by IanLondon