Opentrons/opentrons #2759: Mix displays in step list

howisthisnamenottakenyet, 2018-12-03:
Behavior: mix advanced settings displays its aspirates and dispenses in the sidebar

pantslakz, 2018-12-06:
@howisthisnamenottakenyet ticket should cover following cases:
- mix step
- transfer/consolidate/distribute when user selects mix in advanced setting

how should mix step be represented in sidebar?

IanLondon, 2018-12-06:
Right now, the substeps avoid showing the inner mixes (eg in transfer/consolidate/distribute) by re-runing the timeline with all the `[ ] Mix` checkboxes cleared out - if substeps are in any way affected by mix fields in transfer/consolidate/distribute, that's a bug. I wasn't able to produce that bug a few days ago, it seems like those "inner mixes" are correctly filtered out of the substeps.

But - in Mix step, it shows all the aspirate/dispenses. We can definitely change that but I think it's a feature not a bug?

howisthisnamenottakenyet, 2018-12-06:
@pantslakz @IanLondon Yes this was a bug. Mix advanced setting is not supposed to display in sidebar.

IanLondon, 2019-09-30:
Talked on Slack -- we don't want to do this, we'll revisit substeps/timeline more holistically later on

closed 2019-09-30 by IanLondon