Opentrons/opentrons #2772: Python server commands log: misleading "rate" for aspirate/dispense

IanLondon, 2018-12-03:
## overview

Run App command log (and equivalently, `robot.commands`) is not showing a useful "rate" for aspirate/dispense.

Reported in #2764

## current behavior

This command list is being generated `api/src/opentrons/commands/commands.py` - the definition of `aspirate` here prints the `rate`. The `rate` is not the "speed" but a multiplier (default `1.0`) which scales the `aspirate_speed`.

The JSON executor uses `pipette.set_flow_rate` method for each atomic aspirate/dispense command. This value is not reflected in the `robot.commands()` text that we see in the run log.

## steps to reproduce

In JSON, have a transfer/distribute/consolidate/mix step with `aspirate.params['flow-rate']` set to some number. Upload to Run App and look at the run log on the RUN page.

In Python, call `myPipette.set_flow_rate(aspirate=something, dispense=another_thing)` then do a transfer or something to create some commands. Call `robot.commands()` and look at the command list.

## expected behavior

Perhaps it would be more readable to log: `'Aspirating {volume} uL from {location} at {flow_rate} uL/sec'`? Or at least drop the "rate 1.0" part.

utimcraig, 2018-12-05:
I think 'Aspirating {volume} uL from {location} at {flow_rate} uL/sec' would be exactly a correct and expected thing to do here.

sfoster1, 2020-09-17:
This was actually done by #6189

closed 2020-09-17 by sfoster1