Opentrons/opentrons #2773: PD ignores flow rate settings

IanLondon, 2018-12-03:
Something broke recently in PD: aspirate/dispense flow rates are not being saved in the subprocedures.

Reported in #2764 (from the latest release I assume?) and still true on `edge` b26abddf6b99082591a596a583827bae41f157d1

Example protocol: [slow-asp-then-slow-disp.json.zip](https://github.com/Opentrons/opentrons/files/2641595/slow-asp-then-slow-disp.json.zip) -- open in PD, re-save, and see that the flow rates aren't being set in the `"procedure"` section

Turns out I never implemented them in `step-generation`!

utimcraig, 2018-12-04:
This is really important for my experiment so please fix this one quickly!

ajdittmar, 2018-12-04:
I'm having this issue as well.

utimcraig, 2018-12-05:
Hi @IanLondon how is this doing?  I have an experiment waiting for the feature of having control over aspirate and dispense rates.

utimcraig, 2018-12-06:
Go Ian go!

closed 2018-12-06 by IanLondon

utimcraig, 2018-12-06:
Thanks @IanLondon !  Do you know when this will be released so I can update my opentrons 2.0 ?

commit 92899ea1 (2018-12-05): fix(protocol-designer): finish implementing flow rate in PD

Closes #2773
commit 3f2e86fc (2018-12-06): fix(protocol-designer): finish implementing flow rate in PD

Closes #2773
commit 299ce045 (2018-12-06): fix(protocol-designer): finish implementing flow rate in PD

Closes #2773
PR #2782 (2018-12-06): fix(protocol-designer): finish implementing flow rate in PD
## overview

Closes #2773

## changelog

* finish implementing flow rate across all step types and "inner mixes"

## review requests

- Flow rate field should affect main aspirate/dispense JSON commands within Transfer/Distribute/Consolidate/Mix steps
- Flow rate field should affect aspirate/dispense rate within "inner mixes" of transfer/consolidate (distribute mixes are deprecated but should work too)
- Please test by inspecting the JSON commands (should say `"flow-rate": 1` in the atomic commands in the file)
- Please test on a robo [... cut]