Opentrons/opentrons #2775: Miss Calibration for Labware

simonchih, 2018-12-05:
## overview
Server Version 3.5.1, Opentrons app miss calibration for labware 96-deep-well of slot 7.

## current behavior
Total labware:
96-deep-well in slot 7
tiprack-200ul in slot 2
trough-12row in slot 8

But we can only calibrate:
tiprack-200ul in slot 2
trough-12row in slot 8

Note: 96-deep-well is used while pipette dispense

## steps to reproduce
1. Download python script below:
[Magnetic_test_only_20181204.py](https://drive.google.com/open?id=1lOWz37KxdMALmT3UZYwUQ_Dzfug5rAF2)

2. Open Opentrons GUI App and connect to OT2 via USB

3.  Opentrons GUI App -> PROTOCOL -> Open Magnetic_test_only_20181204.py

4. Opentrons GUI App -> PROCEED TO CALIBRATE
==> Miss calibration for labware 96-deep-well of slot 7.

## expected behavior
App can calibrate all labwares include 96-deep-well of slot 7.

mcous, 2018-12-05:
@simonchih thanks for this report! This bug came up in testing on Monday and now you've saved we having to write a ticket, which I appreciate.

This is a bug with how the robot decides which pipettes and labware are "used" in a given protocol. For some reason, if a labware is dispensed to without an explicit volume, the protocol simulation thinks that labware is "ununsed" and doesn't report it to the app. This is related to #234.

As a workaround, if you add an explicit volume to the dispense (i.e. `p50.dispense(50, p96)`) does the labware show up in the calibration list?

mcous, 2019-06-25:
Oops, this ticket was fixed by #3159. Sorry for leaving it open!

closed 2019-06-25 by mcous