Opentrons/opentrons #2820: Safety Features: Automatic Door Safety Stop (API)

umbhau, 2018-12-13:
As a robot user, I would like my robot to stop automatically when the top cover or door is open, so that I don't injure myself.

## Acceptance Criteria
- [ ] New API feature flag to stop robot and prevent resume when door or top enclosure is open
- Note this only applies to protocol run movement (not calibration, etc.)

mark-nicholas, 2019-07-26:
here is how I enabled the door interlock I hope this helps...

#door interlock
import opentrons, threading
def door_stop():
    opentrons.robot.connect()
    while opentrons.robot.is_connected() is True:
        if robot._driver.read_window_switches() is False:
            opentrons.robot._driver.kill()
            try:
                if robot._driver.read_window_switches() is True:
                    opentrons.robot.connect()
                    opentrons.robot.home()
            except:
                continue

doorstopthread = threading.Thread(target= door_stop, args=())
doorstopthread.daemon = True
doorstopthread.start()

chazmanc, 2019-12-03:
Here's another solution to this problem... @umbhau 

```
from opentrons import labware, instruments, robot
import time
import threading

def door_stop():
    while robot.is_connected() is True and done is False:
        if robot._driver.read_window_switches() is False:
            robot.pause()
            robot._driver.run_flag.wait()
        else:
            time.sleep(2)

done = False

doorstopthread = threading.Thread(target=door_stop, args=(), daemon=True)
doorstopthread.start()

# Protocol goes below

tips = labware.load('opentrons_96_tiprack_300ul', 2)
in_plate = labware.load('biorad_96_wellplate_200ul_pcr', 5)
out_plate = labware.load('biorad_96_wellplate_200ul_pcr', 6)
pip = instruments.P300_Single(mount='left', tip_racks=[tips])

pip.transfer(50, in_plate.wells(), out_plate.wells())

done = True
```

PR #5706 (2020-06-02): feat(api, app): implement automatic door safety stop feature
## overview
This PR adds the automatic door safety stop feature, and enables it if the user has toggled on the `enableDoorSafetySwitch` feature flag. 

closes #2820, closes #2752 

## changelog
- new `enableDoorSafetySwitch` feature flag

- added `doorState` (door open/closed) and `blocked` (if door safety switch is enabled and door is open) to session info

- utilizes the `hardware_event_watcher` in session to monitor hw controller door state changes

- always clears existing session before creating a new one

- used the two added i [... cut]
commit ad94d074 (2020-06-02): feat(api, app): implement automatic door safety stop feature (#5706)

* feat(api, app): implement automatic door safety stop feature

closes #2820, closes #2752
closed 2020-06-02 by ahiuchingau