Opentrons/opentrons #2822: Robot freezes when using protocol from PD

utimcraig, 2018-12-13:
## overview

When I am using a protocol from protocol designer, the robot freezes after the first aspirating/dispense step.  The steps after the first one use the "New pipette tip "never"" option, I'm not sure if that is related to this issue.  This protocol is from the [PD ignores aspirate/dispense speeds](https://github.com/Opentrons/opentrons/issues/2773)

## current behavior

Robot stops working, run timer also stops.  
[image]

The api.log mentions a serial response error.  I also have run this protocol using only USB, USB+Wifi, and Wifi connections and it happens on every type of connection between the computer and the robot.  I'm guessing maybe this serial connection doesn't really have anything to due with the wifi vs usb connection but you guys can help hopefully.

Additionally, after getting this error I have to power cycle the robot to get it to communicate with the computer again.

## steps to reproduce

I tried running either the test.json file (renamed here so I can upload it to github).  The api and serial logs are uploaded here:  
[api.log](https://github.com/Opentrons/opentrons/files/2677840/api.log)
[serial.log](https://github.com/Opentrons/opentrons/files/2677841/serial.log)

[Test-json-renamed to txt for github upload.txt](https://github.com/Opentrons/opentrons/files/2677843/Test-json-renamed.to.txt.for.github.upload.txt)

## expected behavior

The robot should perform the protocol from PD without freezing.

utimcraig, 2018-12-13:
Here is the error from the developer console:
logger.js:43 [app/src\robot\api-client\index.js] warn: Error response from robot
(anonymous) @ logger.js:43
a @ logger.js:32
e.onmessage @ index.js:27
logger.js:51 Object
logger.js:43 [app/src\robot\api-client\index.js] warn: Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 340, in _recursive_update_position
    return _parse_position_response(position_response)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 144, in _parse_position_response
    raise ParseError(msg)
opentrons.drivers.smoothie_drivers.driver_3_0.ParseError: Unexpected response in _parse_position_response: 

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 340, in _recursive_update_position
    return _parse_position_response(position_response)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 144, in _parse_position_response
    raise ParseError(msg)
opentrons.drivers.smoothie_drivers.driver_3_0.ParseError: Unexpected response in _parse_position_response: L:703330305F6D756C74695F76310000000000000000000k

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/server/rpc.py", line 287, in make_call
    self.executor, self.call_and_serialize, func)
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/server/rpc.py", line 277, in call_and_serialize
    call_result = func()
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/api/calibration.py", line 88, in return_tip
    inst.return_tip(home_after=True)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/commands/commands.py", line 371, in decorated
    res = f(*args, **kwargs)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/legacy_api/instruments/pipette.py", line 870, in return_tip
    self.drop_tip(self.current_tip(), home_after=home_after)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/legacy_api/instruments/pipette.py", line 1064, in drop_tip
    return _drop_tip(location)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/commands/commands.py", line 371, in decorated
    res = f(*args, **kwargs)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/legacy_api/instruments/pipette.py", line 1055, in _drop_tip
    self._home_after_drop_tip()
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/legacy_api/instruments/pipette.py", line 1101, in _home_after_drop_tip
    self.robot.poses, safety_margin)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/legacy_api/robot/mover.py", line 77, in fast_home
    safety_margin=safety_margin
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1223, in fast_home
    return self.home(axis=axis, disabled=disabled)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 1187, in home
    self.update_position(default=homed)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 350, in update_position
    DEFAULT_COMMAND_RETRIES)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 347, in _recursive_update_position
    return _recursive_update_position(retries)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 347, in _recursive_update_position
    return _recursive_update_position(retries)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 344, in _recursive_update_position
    raise e
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 340, in _recursive_update_position
    return _parse_position_response(position_response)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 144, in _parse_position_response
    raise ParseError(msg)
opentrons.drivers.smoothie_drivers.driver_3_0.ParseError: Unexpected response in _parse_position_response: 676C655

(anonymous) @ logger.js:43
a @ logger.js:32
e.onmessage @ index.js:28
blob:file:///4eed1a3c-86c3-4d64-b0d4-fd0a5eb14632:14 "session" message: {state: "running", startTime: 1544739385916, lastCommand: {…}, _id: 1882715552}
blob:file:///4eed1a3c-86c3-4d64-b0d4-fd0a5eb14632:14 "session" message: {state: "running", startTime: 1544739385916, lastCommand: {…}, _id: 1882715312}
[... 53 lines cut ...]
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/smoothie_drivers/driver_3_0.py", line 694, in _set_saved_current
    self._send_command(self._generate_current_command())
[... cut ...]
    response = _write_to_device_and_return(command, ack, device_connection)
  File "/data/packages/usr/local/lib/python3.6/site-packages/opentrons/drivers/serial_communication.py", line 68, in _write_to_device_and_return
    device_connection.timeout))
opentrons.drivers.serial_communication.SerialNoResponse: No response from serial port after 1 second(s)

(anonymous) @ logger.js:43
a @ logger.js:32
e.onmessage @ index.js:28
logger.js:47 [app/src\analytics\index.js] debug: Trackable event
logger.js:51 Object
temp1
{type: "@@router/LOCATION_CHANGE", event: {…}}

sfoster1, 2018-12-17:
@utimcraig thanks for this report!

This is a bug with the robot's safety features to catch bugs in its motor controller board. It uses a fixed 30-second timeout. If a command takes more than 30 seconds to complete - so, for a move command, more than 30 seconds to start and stop moving - you get the error you see.

In your protocol, on line 28 you set the flow rate of the pipette to 3 uL/s:
```python
p300.set_flow_rate(aspirate=3, dispense=3)
```
Later, on line 42, you aspirate 160 uL:
```python
p300.aspirate(160, lysate_plate.cols('1'))
```

Because of the flow rate restriction, this takes (160/3) 53.33 seconds, longer than the 30 second timeout, and so the error is raised.;

To be clear, this is not your mistake - we intend to fix this, and will use this issue to track the work to do so. However, as a workaround until then, you have two options:

- Increase your flow rate
- Split your aspirates into smaller segments, in this case always less than 89 uL (to give the system time to accelerate)

On our part, we will add dynamic timeouts for serial commands based on a projection of how long the command will take.

Again, thank you for this report!

utimcraig, 2018-12-17:
Thanks @sfoster1 , it would be nice to not have to worry about a step being under a time limit.  Especially mixing steps.  This kind of aspirate-dispense with low flow rate is critical for phytip  [purifications:](https://phynexus.com/products/proteins/recombinant-protein-phytip-columns/his-tagged-proteins-purification-imac-nickel-nimac-columns/)

I think a lot of people may be interested in this application.  This time limit especially hurts in .mix() operations

utimcraig, 2018-12-17:
@sfoster1 It appears this time limit is not only for single steps, the aspirate+dispense time needs to be less than 30s.    So the example you gave is incorrect, each aspirate/dispense needs to **not** be 89 uL, but rather (89/2) = 44.5 uL.  

I think the best thing is to dynamically calculate a timeout, though this gets complicated because you also have to include head moving times.  Perhaps a heuristic upper limit could be set for the steps taking into account the slowest flow rates, largest volumes, largest moving distances, and head movement speeds. 

For mix() commands I'm not sure how you could set it other than dynamically.  That really is the best option.

utimcraig, 2018-12-19:
I have a bit of a workaround for this issue.  If you use a pipette.delay(seconds=0) in-between aspirate steps it seems to re-start the 30s timer.  So in order to aspirate 100ul at 1ul/sec you can do something like this-

pipette.aspirate()
pipette.delay()
pipette.aspirate()
pipette.delay()

The best solution is still to dynamically calculate the limit on the time of a step though

utimcraig, 2019-01-03:
Hope everyone had a good holiday.  Is anyone working on this bug?  Thanks

utimcraig, 2019-02-11:
Any updates on the dynamic time calculation?

mcous, 2019-02-11:
HI @utimcraig! We're busy working on some core problems in the code base that are preventing us from tackling this problem. You'll see updates in this thread when that work is complete and we have engineering resources available to tackle this one. Thank you for your patience!

PR #3389 (2019-04-30): refactor(api): Split smoothie command ack timeout from motion
Currently, after every command we send an M400, which delays until the
smoothie's queue is empty. We couple this with retrying unacked commands. The
problem is, this couples command acknowledgement with command execution. The
timeouts need to be short enough to be responsive if a command isn't
acknowledged, but short timeouts cause problems with commands that take a long
time - like long motions at low speeds.

Instead, write the command and retry it if not acknowledged, but do not do an
M400 until the command is acknowledged. Once  [... cut]
closed 2019-04-30 by Laura-Danielle