Opentrons/opentrons #2824: Feature request: Opentrons anaconda package

utimcraig, 2018-12-14:
## overview

Anaconda is the most popular data science python platform https://www.anaconda.com/ and it contains an easy to use GUI virtual environment and package management system.  It would be nice to have the correct opentrons api able to be installed via this system.

## current behavior

Currently opentrons does not appear in the anaconda package menu.
[image]

## steps to reproduce

Open anaconda, click on environments, change the dropdown to ALL, and then type opentrons in the search bar.
## expected behavior

Opentrons 1 API and opentrons 2 API should be able to be selected for installation into a virtual environment.

mcous, 2018-12-18:
@utimcraig publishing the protocol API is something we'd very much like to do but is not technically feasible until [a very large refactor / disentaglement](https://github.com/Opentrons/opentrons/milestone/64) of the current robot software monolith is completed. We'll update this thread when we have actual plans around publishing libraries

sfoster1, 2019-02-13:
@utimcraig Good news, we're intending to publish our API on pypi in our next release. I'll update this ticket again when we've done so; it should be in the next couple of weeks. Thanks for your patience!

mcous, 2019-06-25:
Hi @utimcraig, sorry for the lack of update here. We now publish the `opentrons` package to PyPI. Please see the [Anaconda docs](https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-pkgs.html#installing-non-conda-packages) for instructions on how to install a PyPI library into Anaconda via `pip`

closed 2019-06-25 by mcous