Opentrons/opentrons #2834: Robot exposes root access with no password to network

laserson, 2018-12-17:
The OT-2 exposes passwordless root access to the entire network (I believe). Is this intended?

sfoster1, 2018-12-18:
It's intended for troubleshooting. We know it's bad. We have work planned for Q1 and Q2 (but not yet on GH - will update with links to the tickets when they're created) to overhaul the userspace as a whole and move to an actual secure system. We take this very seriously and as a team are quite eager to fix it.

laserson, 2019-01-02:
Following up on this. Will anything break if I change the root password?

mcous, 2019-08-29:
As of 3.11.x, SSH access to the robot uses public key authentication (with no password access). You can add your public key to the robot via HTTP when you are connected via the USB-to-Ethernet connection

https://support.opentrons.com/en/articles/3287453-connecting-to-your-ot-2-with-ssh

closed 2019-08-29 by mcous