Opentrons/opentrons #2835: Need better name for `opentrons-tiprack-300ul`

laserson, 2018-12-17:
It is confusing that the 50 µL pipette is supposed to work with `opentrons-tiprack-300ul`. If it's interchangeable, perhaps consider `opentrons-tiprack`. Or `opentrons-tiprack-50ul-300ul`. Or allow synonymous aliases for a single container.

mcous, 2018-12-18:
Thanks for the feedback @laserson! Would it be correct to generalize your request to "it is confusing and difficult to figure out which tiprack I am supposed to use with a given pipette"?

If I'm not mistaken, that particular tiprack is named that way because the tips themselves have a 300 µl capacity (independent of the capacity of pipette that's using it). As we improve the labware discovery and usage experience - depending on how various pieces of the puzzle enter into it (e.g. documentation, protocol API for requiring a pipette, Protocol Designer GUI) - we may or may not change the labware identifier. However, "I have a pipette and I need to know what tiprack(s) to use" is an experience that will be taken into heavy consideration

laserson, 2018-12-18:
>Would it be correct to generalize your request to "it is confusing and difficult to figure out which tiprack I am supposed to use with a given pipette"?

Sure

> that particular tiprack is named that way because the tips themselves have a 300 µl capacity

I am using TipOne tips because I need filter tips. The 50 µL and 300 µL tips are definitely very different. But regardless, it feels strange to declare a 50 µL pipette to use a container for 300 µL tips. If the overall shape of the rack is the same (which is turns out it is), might as well eliminate the number.

$0.02

mcous, 2018-12-18:
Ah I see! Technically the `opentrons-300ul-tiprack` definition refers _specifically_ to [this tiprack](https://shop.opentrons.com/collections/opentrons-tips/products/opentrons-300ul-tips-racks-9-600-tips) loaded with 300µL tips. Our plans moving forward are to have labware definitions for each specific piece of labware from a specific manufacturer, regardless of geometry similarities.

Are you loading the TipOne tips into the Opentrons racks or are you using TipOne tips? Either way, in the world we're moving to the "correct" thing to do would be to have different definitions for the tiprack holding 50µl TipOne tips and the 300µl TipOne tips.

Obviously we don't have those definitions at this time, so your best move right now _is_ to use the `opentrons-300ul-tiprack` definition, but I'd keep in mind that it's kind of a "hack" until the new labware system is in place

laserson, 2018-12-19:
We used the TipOne tips in the TipOne racks using the printed adapter. Seemed to work ok with the `opentrons-tiprack-300ul`.

mcous, 2019-07-17:
We've just launched an all-new set of labware in the `3.10.0` release and launched the [Labware Library](https://labware.opentrons.com) documentation site. Our goal with the site is to make it clear that a given labware definition is modeled off engineering data for a specific piece (or set of pieces) of labware from a specific manufacturer.

Because the definition of a tip rack (by operational necessity) contains information about the tips themselves, the volume callout in the load name will remain in place.

If you're using TipOne racks with the slot adapter, we have the `tipone_96_tiprack_200ul` definition available, which corresponds to this specific TipOne product: <https://www.usascientific.com/200ul-tipone-stacks.aspx>. Since you're using other volume tips, I'd recommend reaching out to support or filing a ticket for your specific labware

closed 2019-07-17 by mcous

mcous, 2019-07-17:
In the context of *non-filter* tips, the `tipone_96_tiprack_200ul` definition is provided for users who are still using the (non-filter) TipOne tip racks with adapters we used to provide. Since we no longer provide those tip racks, that definition is not included in the library. There are a few definitions like this, and they are called out in [this section](https://docs.opentrons.com/labware.html#deprecated-labware-load-names) of the Python protocol API docs.

For your use case with filter tips, we're working on (but do not yet have) filter tip racks and filter tip rack definitions. To request your labware, please use the form linked in this article:  https://support.opentrons.com/en/articles/3137452-requesting-a-custom-labware-definition. Based on requests and quality of data, we'll use responses to that form to add more library definitions

Thanks for your patience as we've been overhauling how we deal with labware!