Opentrons/opentrons #2836: Support connecting OT-2 directly into wall ethernet port

laserson, 2018-12-17:
For my institution's WiFi, I must use my personal login credentials, which requires me to regularly update my password. Alternatively, if I could connect the OT-2 directly into the wall, it would get a stable IP address with no authentication required. Is it possible to somehow physically connect the robot through its exposed USB port into the wall (without taking off the robot case to expose the RJ45 on the RPi).

mcous, 2018-12-17:
@laserson thanks for the request! We are actively working on supporting a regular DHCP'd ethernet connection for the OT-2, but we're still figuring out what our official path forward will be.

It seems like you may know this, but if not, the exposed USB port on the frame is one end of a USB-to-Ethernet adapter inside the frame that is plugged into the Pi's Ethernet port. The Pi's Ethernet interface is configured to be link-local only, so even if you did open the frame, using that Pi's Ethernet port for a DHCP network wouldn't work.

While not ideal, if you plug _another_ USB-to-ethernet adapter into one of the exposed USB ports on the Pi, the Pi should pick it up as a regular network adapter and connect to the network, where everything _should_ just work. The app will erroneously report the robot as being connected over Wi-Fi because it makes the assumption that any robots coming over the LAN are Wi-Fi connected, but everything should work aside from that!

Keep in mind that the app relies on mDNS to discover robots over LAN, so if mDNS is unable to traverse your network(s), a robot wired into Ethernet may not show up in the app automatically and would require its IP address to be added to the app's configuration file as detailed in the #2419 thread

laserson, 2018-12-18:
That's incredibly helpful! What is the actual machine/OS inside the robot? (I just want to make sure I get an adaptor that works out-of-the-box.)  I'll report back if it works for me, in which case you should definitely add this to the docs.

mcous, 2018-12-18:
The OT-2 runs on a Raspberry Pi 3 Model B running [Resin OS 2.7.5](https://www.balena.io/os/) (since renamed Balena) as its host OS (with our application running in an Alpine Linux container). `uname -a`  for the host OS, which deals with network adapters:

```
Linux 9eb674d 4.9.50 #2 SMP Wed Nov 1 16:03:39 UTC 2017 armv7l armv7l armv7l GNU/Linux
```

Anecdotally, all the USB 3.0 USB-to-Ethernet adapters I've tried have worked. I use a Satechi one that I picked up from I-don't-know-where for development and it's compatible

laserson, 2019-01-02:
I can confirm this works for me (with an Anker gigabit adaptor). I connected using USB, then ssh'd into the robot using the IP address specified in the app under the "Connectivity" section for "USB", then used `ifconfig` to find the IP address of the ethernet adaptor, then added it to the `config.json` in `~/Library/Application Support/Opentrons`, then restarted the app.

mcous, 2019-08-29:
I _think_ this might actually be possible now with 3.11.x? I'm going to investigate today and update this thread accordingly

mcous, 2019-08-29:
@laserson alrighty I've confirmed that if you plug an Ethernet cable directly into the RPi's RJ-45 port and there's a DCHP server on the other end, the Pi will connect to that network properly.

Unfortunately, with the OT-2 hardware as it is, it's not possible to get to that port without opening the robot. You might be able to take off the side panel (rather than the box with the Pi) to get to the USB-to-Ethernet adapter and highjack that cable.

I'm very much of the _personal_ opinion that we should be exposing an RJ-45 port on the robot rather than this "fake" USB port. I'll be continuing to push on this issue with the product / hardware teams, but it is a larger decision (and there are ease of use arguments to be made in favor of "USB"), so I can't guarantee anything.

Given that the Ethernet port itself will behave if you get access to it, and given that there is a workaround (via another USB-to-ethernet adapter) available if you cannot get access to it, I'm going to close this ticket. I'll post an update and/or re-open if there's any movement on the hardware front

laserson, 2019-08-29:
Thanks @mcous. As I reported before, the adaptor you suggested works for us, and we simply plug the robot into the wall. Would be great to at least put a "protocol" up on the website about how to do this. It's especially helpful at places like where I work (large hospital) where setting up the WiFi is a hassle or impossible.

mcous, 2019-08-29:
Good call, I'll send get a support article drafted up and posted

mcous, 2019-09-06:
Update: we believe 3.11.x may have broken hotplug capabilities on external USB-to-Ethernet adapters. Re-opening this ticket until that is resolved. Workaround follows:

1. SSH into the robot per the instructions at https://support.opentrons.com/en/articles/3287453-connecting-to-your-ot-2-with-ssh via the robot’s **built-in** Ethernet-to-USB port
2. Once in SSH, verify that the the **external** USB-to-Ethernet dongle is showing up in `nmcli device`
```
# nmcli device
DEVICE  TYPE      STATE         CONNECTION
eth0    ethernet  connected     wired-linklocal
eth1    ethernet  disconnected  --
wlan0   wifi      disconnected  --
lo      loopback  unmanaged     --
```
3. Create a new auto (DHCP) connection for that dongle interface
```
# nmcli connection add type ethernet con-name dongle-dhcp ifname eth1
Connection 'dongle-dhcp' (bed0ec0c-f7dc-415c-abb0-0cb8d229b11a) successfully added.
```
4. Verify the new connection is up and running
```
# nmcli connection
NAME             UUID                                  TYPE      DEVICE
dongle-dhcp      bed0ec0c-f7dc-415c-abb0-0cb8d229b11a  ethernet  eth1
wired-linklocal  493aa3ef-7814-3000-b778-ac53348b2113  ethernet  eth0
static-eth0      55a8a985-ae9a-4541-b4fb-f3729194d22f  ethernet  --
wired            fb183d70-856c-3a09-b89f-78ee5ce5080e  ethernet  --
```

The app should now show that the robot “The robot is connected to a network and has full access to the Internet.”

closed 2019-09-10 by sfoster1

mcous, 2019-10-31:
Hey @theosanderson, DHCP functionality on the RPi's RJ-45 port is available in the current release. As mentioned in the thread above, if getting to the built-in port is difficult, plugging a USB-to-Ethernet adapter into one of the Pi's USB ports will also work to connect it to a wired network

Funnily enough, I just came out of a meeting about the exposed Ethernet port. We don't have a timeline or anything, but there's pretty broad consensus on our side to make that change happen. Your +1 is definitely noted

theosanderson, 2019-10-31:
Thanks @mcous !