Opentrons/opentrons #2843: Can't cancel a protocol, it blocks in cancel status

simonchih, 2018-12-19:
## overview
When protocol is running in delay status, we can't cancel the protocol immediately, then it blocks in cancel status.

## current behavior
Protocol blocks in cancel status, we can't see "RESET RUN" in app.

## steps to reproduce
1. Open Opentrons GUI App and connect via USB
2. Upload protocol below:
```
from opentrons import labware, instruments, modules

p10rack = labware.load('tiprack-10ul', '1', 'p20-rack')

module = modules.load('tempdeck', '10')
f = labware.load('96-flat', '10', share=True)

p10 = instruments.P10_Single(
    mount="left",
    tip_racks=[p10rack]    
)

module.set_temperature(95)
module.wait_for_temp()
p10.delay(3600)
module.deactivate()
```
3.  Wait 10~30 minutes
4. Cancel protocol by app's button
==> Protocol blocks in cancel status, we can't see "RESET RUN" in app.

## expected behavior
Cancel protocol successfully and show "RESET RUN" button.

mcous, 2019-08-29:
We need to investigate whether this is still an issue with 3.11.x

mcous, 2019-08-29:
I ran the following protocol and was able to cancel during the delay. I don't know when exactly we fixed this, but as of 3.11.x it is resolved

```py
from opentrons import robot, labware, instruments

plate = labware.load("biorad_96_wellplate_200ul_pcr", slot="2")

reservoir = labware.load("usascientific_12_reservoir_22ml", slot="1")

tiprack = labware.load("opentrons_96_tiprack_300ul",
                       slot="5", label="tiprack under test")

pipette = instruments.P300_Multi(mount="right", tip_racks=[tiprack])

pipette.pick_up_tip(tiprack.wells('A1'))
pipette.delay(seconds=60)

pipette.aspirate(50, reservoir.wells('A1'))
pipette.dispense(50, plate.wells('A1'))
pipette.aspirate(50, reservoir.wells('A1'))
pipette.dispense(50, plate.wells('A1'))

pipette.return_tip()
```

closed 2019-08-29 by mcous