Opentrons/opentrons #2893: Wait for enter key press crashes on upload to robot

utimcraig, 2019-01-08:
## overview
When uploading a python script that requires a user to press a button to continue the protocol the upload to the robot crashes.  

## current behavior
Currently I am uploading a python program to the robot through the OT2 app that contains the following:
```
robot.comment("Hello, please check that you have put 1M Tris pH 9.0 into the elution column then press ENTER")
try:
    input("Press ENTER to continue")
except SyntaxError:
    pass

```
When I run this in jupyter notebook it works correctly and asks me to press the enter key before running the rest of the program.  When try to upload this program to the robot it gets stuck on the upload screen.  My guess is that the software is trying to run the program and is getting stuck on the required enter key, which is not being "pressed" on the robot.

[image]

The program can currently not be run on the robot.  I would like to have a message appear on the screen's run notes saying 

"Hello, please check that you have put 1M Tris pH 9.0 into the elution column then press ENTER"

at which point the user would press the enter key on their computer and the program would continue

## steps to reproduce

Uploading the following file will result in a "Upload in progress" that lasts forever:
[Phytips_ProteinA-v1_py_renamed_to_txt.txt](https://github.com/Opentrons/opentrons/files/2738351/Phytips_ProteinA-v1_py_renamed_to_txt.txt)

## expected behavior

 I would like to have a message appear on the screen's run notes saying 

"Hello, please check that you have put 1M Tris pH 9.0 into the elution column then press ENTER"

at which point the user would press the enter key on their computer and the program would continue

Is there a better way to do this type of thing from within the opentrons api?

mcous, 2019-01-09:
Hello @utimcraig! I believe you are looking for the [`robot.pause`](https://docs.opentrons.com/api.html#opentrons.legacy_api.robot.Robot.pause) API method:

```py
robot.pause(
    msg="Check that you have put 1M Tris pH 9.0 into the elution column"
)
```

This message will then show up in the run log, and when the robot hits that line it will pause operation, which you can then resume by clicking "Resume" in the app.

[image]

Unfortunately, we have a known bug with run log highlighting (#2047) which means the step with the message may not be the step that is highlighted when the robot pauses. The underlying cause is way more complicated than it should be, which is why it's taking us so long to fix the situation, but please do know that even though the log highlighting is incorrect the robot will actually pause in the correct place.

closed 2019-01-09 by mcous

utimcraig, 2019-01-09:
@mcous Is there a resume button during the upload?  The problem I'm having is that the program upload to the robot does not complete.

mcous, 2019-01-09:
You cannot include `input` in your protocol if you want to run via app upload. The upload is stalling because the simulation is waiting for you to to enter something into the (non-existent) robot keyboard.

You should replace:

```py
robot.comment("Hello, please check that you have put 1M Tris pH 9.0 into the elution column then press ENTER")
try:
    input("Press ENTER to continue")
except SyntaxError:
    pass
```

With

```py
robot.pause("Hello, please check that you have put 1M Tris pH 9.0 into the elution column then click RESUME")
```

...then your protocol will upload/simulate successfully. As noted, the incorrect step will be highlighted in the run log once the robot actually pauses, but that is a separate issue that does not affect actual protocol execution.
Then the simulation will complete successfully.