Opentrons/opentrons #2941: Pipette Config: Configs Form

umbhau, 2019-01-20:
As an OT-2 user, I would like to be able to input pipette config values via the Run App

## Acceptance Criteria
- [ ] Run App form to set pipette config values (see field list, below)
- [ ] Individual fields should support:
 - [ ] Labels
 - [ ] Min, Max, and Default value guidelines
 - [ ] Units (in field)
- [ ] Form should have a 'reset' button to return all values to their defaults (at the UI level)
- [ ] Fields should have tooltips
- [ ] Form should display a progress/pending icon on save

## Design
- TBD

## Resources
https://docs.google.com/spreadsheets/d/1PeR0UNynMRTnvSgpnBODozrtI1rgzT5MlAuFdMugfSI/edit?usp=sharing
(note: still need labels, tooltips)

## Implementation Notes
- [ ] Mock response; BE wiring in a separate story
- [ ] Form validations in a separate story

pantslakz, 2019-02-01:
@Kadee80 note that the "assistance" text in warning should trigger an intercom chat window. not sure if that requires its own ticket.

pantslakz, 2019-02-01:
@umbhau we require copy for the tooltips. we put the range in the tooltip - but note that I'm not sure if we've ticketed out how we plan to keep that info dynamic as ranges change.

PR #3089 (2019-02-27): refactor: Render pipette config form fields from data
## overview

This PR closes #2941 by rendering a Pipette Config Form using on mock data. I'm cutting this PR off before wiring before it gets too big.

Mock data is copy/pasted from a postman response from #3070 .

## changelog

- refactor: Render pipette config form fields from data

## review requests

```
make -C app dev OT_APP_DEV_INTERNAL__NEW_PIPETTE_CONFIG=1
```
This PR does:
- [ ] Render pipette config modal
- [ ] Render pipette display name in title
- [ ] Render labeled/grouped form fields from hard coded data
- [ ] Render default  [... cut]
commit 239dda43 (2019-02-27): refactor: Render pipette config form fields from data (#3089)

closes #2941
closed 2019-02-27 by Kadee80